kubeedge/mappers-go #8: Modbus go mapper error

beyondyinjl2, 2021-01-15:
sudo docker build -t modbusmapper:v1.0 ./pkg/modbus

Sending build context to Docker daemon  52.74kB
Step 1/6 : FROM ubuntu:16.04
 ---> 9499db781771
Step 2/6 : RUN mkdir -p kubeedge
 ---> Running in b680bf86b7e1
Removing intermediate container b680bf86b7e1
 ---> e88ad3020b46
Step 3/6 : COPY ./modbus kubeedge/
COPY failed: stat /var/lib/docker/tmp/docker-builder170465191/modbus: no such file or directory

not modbus dir

applea9, 2021-01-26:
you should compile & build modbus mapper brefore you build docker image. please follow the below steps:
```
make modbusmapper
make modbusmapper_image
```

RyanZhaoXB, 2022-03-29:
please build a binary file for the modbus mapper first.