kubeedge/mappers-go #14: Create mapper template

sailorvii, 2021-04-08:
Create a template for a customized mapper to ease the mapper development.

sailorvii, 2021-04-09:
Target:
The user specifies the new mapper's name, then automatically creates a folder accordingly to the name. The User needs to write several functions, then the mapper works.

Detailed design:
1. Template files.
The file tree is like that:
- <nameOfNewMapper>
   - main.go
   - Dockerfile ---- Combine building code and container image.
   - deployment.yaml
   - config.yaml  
   - driver     ---- This is for reading/writing devices.  
     - client.go    -----  Fill in the functions like getting register/setting register.
   - device    ---- Device management. Such as reading the device configuration and create the data structure accordingly, connecting the devices, and reading/writing device registers.
     - device.go  ---- Almost need not much change except the initialization of driver client and customized visitor parsing.
     - devstatus.go
     - twindata.go
  - configmap
    - type.go   ---- Add the mapper-specific data structure.
    - parse.go
  - globals  ---- Global definitions.
    - globals.go
2. Tool to create the code.
   Use a shell script to get inputted mapper name, copy the template files to a new folder named as input. Replace the related file name, also replace some characters as input.

ttlv, 2021-04-12:
well, this template is just for some protocols which are suppotted C/S,if some protocols are not suppotted C/S,the file client.go is useless.

sailorvii, 2021-04-13:
If not the C/S structure, the client.go is also for getting/setting the device. For example, in modbus rtu mode, the client is to open serial devices and get/set them.

PR #41 (2021-06-01): Add mapper template and refine the make method.
What type of PR is this?
/kind enhancement

What this PR does / why we need it:
Make things easy to develop a new mapper.

Which issue(s) this PR fixes:
Fixes #14
closed 2021-06-01 by kubeedge-bot