kubeedge/mappers-go #36: why mapper Twin structure has PropertyVisitor field

gy95, 2021-04-26:
why mapper `Twin` structure has `PropertyVisitor` field, on the cloud side, this `Twin` structure only has three fields except `PropertyVisitor`
```
type Twin struct {
	PropertyName string `json:"propertyName,omitempty"`
	PVisitor     *PropertyVisitor
	Desired      DesiredData  `json:"desired,omitempty"`
	Reported     ReportedData `json:"reported,omitempty"`
}
```

sailorvii, 2021-04-27:
This element is a pointer used for finding the visitor directly. It's not in the configmap or json files.

gy95, 2021-04-27:
> This element is a pointer used for finding the visitor directly. It's not in the configmap or json files.

Got it, thanks a lot

closed 2021-04-27 by gy95