kubeedge/mappers-go #45: Unable to  build bluetooth mapper image using the new mapper make template

akw123, 2021-06-14:
What happened and what you expected to happen: 
Since version 1.7, the new mapper template to build bluetooth image is not working.

How to reproduce it (as minimally and precisely as possible):
Run make mapper bluetoothmapper_image b ARM-true

Anything else we need to know?:
Error log:
bluetoothmapper_image b
# make mapper bluetoothmapper_image b
downloading dependencies for mapper bluetoothmapper_image...
tidying
vending
...done
fmt and linting mapper bluetoothmapper_image...
WARN [runner] The linter 'golint' is deprecated (since v1.41.0) due to: The repository of the linter has been archived by the owner.  Replaced by revive.
driver/client.go:45:3: undeclared name: `klog` (typecheck)
                klog.Errorf("New device error: %v", err)
                ^
driver/client.go:62:5: undeclared name: `klog` (typecheck)
                                klog.Errorf("Write characteristic error %v", err)
                                ^
driver/client.go:26:2: "k8s.io/klog/v2" imported but not used (typecheck)
        "k8s.io/klog/v2"
        ^
device/device.go:45:3: undeclared name: `klog` (typecheck)
                klog.V(1).Info("Visit readonly characteristicUUID: ", visitorConfig.CharacteristicUUID)
                ^
device/device.go:28:2: "k8s.io/klog/v2" imported but not used (typecheck)
        "k8s.io/klog/v2"
        ^
device/devicestatus.go:20:2: "k8s.io/klog/v2" imported but not used (typecheck)
        "k8s.io/klog/v2"
        ^
../../../../../../snap/go/7736/src/runtime/cgo/cgo.go:34:8: could not import C (cgo preprocessing failed) (typecheck)
import "C"

Environment: Ubuntu 20.10

akw123, 2021-06-14:
Is there a user guide for the new mapper template in 1.7.0 ?

sailorvii, 2021-06-16:
which folder do you use for bluetooth? There's only a folder name "ble".
What I got:
make mapper ble ARM=true
ble
# make mapper ble build
downloading dependencies for mapper ble...
tidying
vending
...done
fmt and linting mapper ble...
...done
crossed packaging for linux/arm
building linux/arm
...done

sailorvii, 2021-06-16:
> Is there a user guide for the new mapper template in 1.7.0 ?

Sorry for the late instruction. I'm working on a new customized mapper and want to use it as an example in the guide. It'll be done in one or two weeks.

akw123, 2021-06-16:
Thanks @sailorvii  make ble I'm able to build.
Didn't know the mapper name has to be the folder name. 

A guide to use the mapper template and updated sample CRDs would be helpful.

I tried to build with 1.6.0
However I noticed there is an issue here with 1.6.0
My devicemodel name is android-ble-simulator, and the watcher is not able to scan the device because it strips the '-' for the device localname look up.

With version 1.7.0 is uses Mac address instead?
Is this mandatory with 1.7.0?

sailorvii, 2021-06-17:
@ttlv Could you help with this issue? Thanks.

ttlv, 2021-06-17:
> Thanks @sailorvii make ble I'm able to build.
> Didn't know the mapper name has to be the folder name.
>
> A guide to use the mapper template and updated sample CRDs would be helpful.
>
> I tried to build [...]

Is there have some problems when using Mac address to connect ble devices?

akw123, 2021-06-17:
> Is there have some problems when using Mac address to connect ble devices?

I have not tried with 1.7 yet. 
But I think it make sense with Mac.
Using device name is not unique and tightly coupled with device model crds. 
Somehow render it not reusable if I have another device with a different name but same model.
IMO it should be part of the device instance configuration. 

I'll give it go once @sailorvii  has uploaded the guide. Thanks for support. 

/close

closed 2021-06-17 by kubeedge-bot