kubeedge/mappers-go #71: a proposal for gige-sdk

WangQing-WUT, 2022-05-10:
Hi, guys. We have make a proposal for gige-sdk, which is a GigE mapper based on [mapper-sdk-go](https://github.com/QiQi-OvO/mappers-go/tree/main/mapper-sdk-go) and supports GigE Vision protocol cameras access.

More details about gige-sdk: [gige-sdk README.md](https://github.com/WangQing-WUT/mappers-go/blob/main/mappers/gige-sdk/README.md)

We can discuss more details about this proposal and welcome to make suggestion for gige-sdk, thanks.

**What would you like to be added/modified:**
We want to use the generic camera protocol genicam protocol as the driver of the GigE mapper to access cameras from different manufacturers, so there is no need to adapt the driver layer for cameras from different manufacturers and reduce customized development. At the same time, we want to develop GigE mapper based on [mapper-sdk-go](https://github.com/QiQi-OvO/mappers-go/tree/main/mapper-sdk-go) and encourage other people to use this new mapper-sdk to help us develop a new mapper more easily.

**Why we need it:**
At present, KubeEdge mapper does not have a mapper suitable for GigE vision cameras access. At the same time, there is no good solution for cameras access from different manufacturers, which can not meet the needs of users for mapper.

RyanZhaoXB, 2022-06-16:
sorry i can not get your `gige-sdk README.md`. please update the link. thanks :)

WangQing-WUT, 2022-06-16:
> sorry i can not get your `gige-sdk README.md`. please update the link. thanks :)
I am very sorry that the project structure adjustment caused the link to become invalid, and the link has been updated in the comments.Thanks for your review :)

closed 2022-09-26 by kubeedge-bot