kubeedge/mappers-go #106: mapper映射器

zzzzzzyzz, 2023-07-20:
Can Kubeedge currently map a machine node through the current mapper

QiQi-OvO, 2023-07-21:
I think you can abstract a machine node into a device, use CRD to describe the attributes of your node. Your idea is very interesting, welcome to continue providing some details. :)

zzzzzzyzz, 2023-07-21:
I have considered this approach, but I am uncertain if the nodes support CRDs. Additionally, there are differences in the way devices and nodes operate

QiQi-OvO, 2023-07-21:
CRD has some custom fields that allow developers to customize thier device, u can find some example in [this](https://github.com/kubeedge/mappers-go/tree/main/build/crd-samples/devices), for example [random-device-instance.yaml](https://github.com/kubeedge/mappers-go/blob/main/build/crd-samples/devices/random-device-instance.yaml) and [random-device-model.yaml](https://github.com/kubeedge/mappers-go/blob/main/build/crd-samples/devices/random-device-model.yaml) implemented a virtual device that generates random numbers. For another question, I think that u can establish a server on the node, which collects indicators that need to be reported to kubeedge. The mapper can obtain and change the corresponding data through the `GET/POST` method. You can attend our community meeting every Wednesday afternoon and we can discuss more details. Of course, if you have any questions, you can continue to communicate with me here.

zzzzzzyzz, 2023-07-21:
Alright, thank you very much.