kubeedge/mappers-go #108: After the modbus device connects to the mapper, the container of the mapper fails.

NicoleLeeuu, 2023-07-31:
The Modbus Mapper is running normally.  After the device is delivered, the container of the mapper fails.
[image]
View the mapper log. The command output is "panic: runtime error: invalid memory address or nil pointer dereference".
[image]
Locate the problem. It was found that the payload in the Run() function, located in the mappers/modbus-dmi/device/twindata.go file, did not add the "Expected" data section. This results in an error when accessing msg.Twin.Expected when executing the parse.ConvMsgTwinToGrpc(msg.Twin) function.
Is this a bug in the process of design and implementation?  If so, please leave bugfix matters to me.

QiQi-OvO, 2023-08-01:
Yes, this is a bug that has been fixed by two PRs(https://github.com/kubeedge/mappers-go/pull/99, https://github.com/kubeedge/mappers-go/pull/105), but they haven't been merged into the code repository yet.

cl2017, 2023-08-18:
> Yes, this is a bug that has been fixed by two PRs(#99, #105), but they haven't been merged into the code repository yet.

Let's review these PRs again and merged it into master

closed 2023-08-18 by NicoleLeeuu